Bodmer/TFT_ILI9341 #1: F_AS_T option

Bodmer, 2015-11-12:
Do not comment out the #define F_AS_T line in the User_Setup.h file as the slower SPI access methods currently have a bug.

Bodmer, 2015-11-27:
Fixed and function memory footprint reduced.

fmaimon, 2015-11-29:
Why leave this option altogether? Is there an advantage of using the slower option?

Bodmer, 2015-11-30:
This is a legacy from earlier versions of the library, the aim was to permit the library to be used on SAM or ARM (e.g. DUE or Teensy) which would not work with the F_AS_T option. 

For AVR processors (Atmega328, ATmega2560, ATmega 32u4 etc)  there is no advantage of not using it as th egraphics plotting will be slower.  AS I use the AVR processors I have not got around to fixing an issue when F_AS_T is not use because I have been focusing on other features that I need for my own projects.

Bodmer, 2015-11-30:
F_AS_T option is still on my list to be fixed!

closed 2015-11-30 by Bodmer