Bodmer/TFT_ILI9341 #2: Why this library should be preferable to Adafruit's one?

q2dg, 2015-11-15:
In other words...what are the changes you have made?

Thanks!

Bodmer, 2015-11-16:
See the linked Instructable in the description and try the examples:
http://www.instructables.com/id/Arduino-TFT-display-and-font-library/
In summary:
+ 3 to 5 times faster for graphics drawing on AVR processors (clears screen in 174ms)
+ nice proportional fonts and formatting/justification functions
+ larger fonts are Run Length Encoded to save FLASH
+ fonts can be easily included/excluded at compile time

closed 2015-11-16 by Bodmer

q2dg, 2015-11-16:
Thanks for your answer.
Have you thought about merging these changes into Adafruit's repository?

Bodmer, 2015-11-16:
I think the changes are too drastic and restrictive as the code is heavily optimised for the AVR processors so almost every function has been modified for this purpose, and hence the choice to keep the library separate.

q2dg, 2015-11-16:
Ok, understood!
Thanks a lot!