Bodmer/TFT_ILI9341 #4: Cleaning up reported minor errors

oldmanegan, 2016-07-28:
Instead of me branching your to eliminate these, I was hoping you might be able to resolve these or add comments to ignore.
Errors reported by Arduino IDE on compile (in verbose mode):
/Volumes/Docs HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.cpp: In member function 'void TFT_ILI9341::writecommand(uint8_t)':
/Volumes/Docs HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.cpp:120:8: warning: unused variable 'spsr' [-Wunused-variable]
   byte spsr = SPSR;// We need this here for some reason...
        ^
/Volumes/Docs HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.cpp: In member function 'void TFT_ILI9341::setAddrWindow(int16_t, int16_t, int16_t, int16_t)':
/Volumes/Docs HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.cpp:974:8: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   if(x1!=win_xe) {
        ^
/Volumes/Docs HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.cpp:989:8: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   if(y1!=win_ye) {
        ^
/Volumes/Docs HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.cpp: In member function 'void TFT_ILI9341::drawPixel(uint16_t, uint16_t, uint16_t)':
/Volumes/Docs HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.cpp:1052:10: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   if ((x >= _width) || (y >= _height)) return;
          ^
/Volumes/Docs HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.cpp:1052:27: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   if ((x >= _width) || (y >= _height)) return;
                           ^
/Volumes/Docs HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.cpp: In member function 'void TFT_ILI9341::fastPixel(uint16_t, uint16_t, uint16_t)':
/Volumes/Docs HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.cpp:1102:10: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   if ((x >= _width) || (y >= _height)) return;
          ^
/Volumes/Docs HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.cpp:1102:27: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   if ((x >= _width) || (y >= _height)) return;
                           ^
/Volumes/Docs HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.cpp: In member function 'virtual size_t TFT_ILI9341::write(uint8_t)':
/Volumes/Docs HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.cpp:1679:50: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     if (textwrap && (cursor_x + width * textsize >= _width))
                                                  ^
/Volumes/Docs HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.cpp: In member function 'int TFT_ILI9341::drawChar(unsigned int, int, int, int)':
/Volumes/Docs HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.cpp:1747:30: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     if (x + width * textsize >= _width) return width * textsize ;
                              ^
/Volumes/Docs HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.cpp:1751:25: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
       for (int i = 0; i < height; i++)
                         ^
/Volumes/Docs HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.cpp:1793:25: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
       for (int i = 0; i < height; i++)
                         ^
/Volumes/Docs HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.cpp:1882:20: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
             if (px >= (x + width * textsize))
                    ^
/Volumes/Docs HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.cpp:1837:28: warning: unused variable 'tpy' [-Wunused-variable]
       int px = 0, py = pY, tpy = pY; // To hold character block start and end column and row values
                            ^
/Volumes/Docs HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.cpp: In member function 'int TFT_ILI9341::drawString(char*, int, int, int)':
/Volumes/Docs HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.cpp:1986:19: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     if (poX+cwidth>_width)   poX = _width - cwidth;
/Volumes/Docs HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.cpp:1988:20: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     if (poY+cheight>_height) poY = _height - cheight;
/Volumes/Docs HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.cpp: In function 'drawNumber.constprop':
/Volumes/Docs HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.cpp:828:18: warning: 'widthtable' may be used uninitialized in this function [-Wmaybe-uninitialized]
     str_width += pgm_read_byte( widthtable + uniCode); // Normally we need to subract 32 from uniCode

Bodmer, 2016-07-30:
Fair comment, I don't use verbose mode :-)

These warnings are all expected due to legacy changes and lack of type casting rigour. I have not encountered any negative effects of these warnings but I will sort them out on the next update.

Bodmer, 2016-07-31:
Next update has been done. These warnings have been eliminated.

closed 2016-07-31 by Bodmer

oldmanegan, 2016-07-31:
Now have new errors and three older ones I had gotten around:
In file included from /Volumes/Docs HD/Electronic Projects and
Info/Arduino/Projects/Egan Smart Home/Sprinkler
Project/Sketches/Sprinkler_3.7.1/Sprinkler_3.7.1.ino:16:0:
/Volumes/Docs
HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.h:369:20: error:
redeclaration of 'int16_t TFT_ILI9341::cursor_x'
            cursor_x, cursor_y, padX;
                    ^
/Volumes/Docs
HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.h:366:12: note:
previous declaration 'int16_t TFT_ILI9341::cursor_x'
   int16_t  cursor_x, cursor_y, win_xe, win_ye, padX;
            ^
/Volumes/Docs
HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.h:369:30: error:
redeclaration of 'int16_t TFT_ILI9341::cursor_y'
            cursor_x, cursor_y, padX;
                              ^
/Volumes/Docs
HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.h:366:22: note:
previous declaration 'int16_t TFT_ILI9341::cursor_y'
   int16_t  cursor_x, cursor_y, win_xe, win_ye, padX;
                      ^
/Volumes/Docs
HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.h:369:32: error:
redeclaration of 'int16_t TFT_ILI9341::padX'
            cursor_x, cursor_y, padX;
                                ^
/Volumes/Docs
HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.h:366:48: note:
previous declaration 'int16_t TFT_ILI9341::padX'
   int16_t  cursor_x, cursor_y, win_xe, win_ye, padX;
                                                ^
/Volumes/Docs
HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.h:374:18: error:
redeclaration of 'uint16_t TFT_ILI9341::_width'
   uint16_t _width, _height, // Display w/h as modified by current rotation
                  ^
/Volumes/Docs
HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.h:368:12: note:
previous declaration 'int16_t TFT_ILI9341::_width'
   int16_t  _width, _height, // Display w/h as modified by current rotation
            ^
/Volumes/Docs
HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.h:374:27: error:
redeclaration of 'uint16_t TFT_ILI9341::_height'
   uint16_t _width, _height, // Display w/h as modified by current rotation
                           ^
/Volumes/Docs
HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.h:368:20: note:
previous declaration 'int16_t TFT_ILI9341::_height'
   int16_t  _width, _height, // Display w/h as modified by current rotation
                    ^
/Volumes/Docs
HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.h:375:21: error:
redeclaration of 'uint16_t TFT_ILI9341::textcolor'
            textcolor, textbgcolor, fontsloaded, addr_row, addr_col;
                     ^
/Volumes/Docs
HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.h:371:12: note:
previous declaration 'uint16_t TFT_ILI9341::textcolor'
   uint16_t textcolor, textbgcolor, fontsloaded;
            ^
/Volumes/Docs
HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.h:375:34: error:
redeclaration of 'uint16_t TFT_ILI9341::textbgcolor'
            textcolor, textbgcolor, fontsloaded, addr_row, addr_col;
                                  ^
/Volumes/Docs
HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.h:371:23: note:
previous declaration 'uint16_t TFT_ILI9341::textbgcolor'
   uint16_t textcolor, textbgcolor, fontsloaded;
                       ^
/Volumes/Docs
HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.h:375:47: error:
redeclaration of 'uint16_t TFT_ILI9341::fontsloaded'
            textcolor, textbgcolor, fontsloaded, addr_row, addr_col;
                                               ^
/Volumes/Docs
HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.h:371:36: note:
previous declaration 'uint16_t TFT_ILI9341::fontsloaded'
   uint16_t textcolor, textbgcolor, fontsloaded;

As well these I had commented out on my own:
In file included from /Volumes/Docs HD/Electronic Projects and
Info/Arduino/Projects/Egan Smart Home/Sprinkler
Project/Sketches/Sprinkler_3.7.1/Sprinkler_3.7.1.ino:16:0:
/Volumes/Docs
HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.h:365:1: error:
expected unqualified-id before '<<' token
 <<<<<<< Updated upstream
 ^
/Volumes/Docs
HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.h:367:1: error:
expected unqualified-id before '==' token
 =======
 ^
/Volumes/Docs
HD/pop/Documents/Arduino/libraries/TFT_ILI9341/TFT_ILI9341.h:372:1: error:
expected unqualified-id before '>>' token
 >>>>>>> Stashed changes
 ^

Bodmer, 2016-08-01:
It looks like you have multiple copies of the library or have made edits that cause these errors.

I can't reproduce these errors and I have tried a few of the examples provided.

Note that the header file in the master copy has changed, it looks like you are not using the latest copy.

oldmanegan, 2016-08-01:
Ah! Github Desktop must not have replaced the User file as I had edited it
to have the right pins for the 1284P implementation. I'll download the
current full Gibub zip and try again.

oldmanegan, 2016-08-01:
Yes, verified as clean. Thanks!