Bodmer/TFT_ILI9341 #5: SetTextColor with just ohne Parameter

Pin666, 2016-07-30:
Hi bodmer,

First of all, great library. Highly appreciate your work. I use it together with a second SPI device. It's a cooking hacks RS485 board to communicate via modbus RTU with a VFD ( Hitachi L200). Everything works fine, just encountered two strange thinks. 
1.) I can not use setTextColor with just one parameter. If I do my second SPI device will not work any longer. I need to use the two param version setTextColor (forecolor, back color). Then everything works fine. It's not really an issue for me. Just wanted to let you know. Maybe YOU know why. 
2.) I was not able to use transactions in your library. If I enable it in user_setup.h nothing works. Luckily I was able so setup my two SPI devices with the same SPI parameters because

Bodmer, 2016-07-30:
Thanks for your feedback.

Re.1 It looks like a bug in the library that is leaving the chip select low. I will do some tests to check this and/or find why this is happening. 

Re. 2. Mmm,  looks like I have broken the transaction support. I will investigate this too.

Thanks for reporting these issues.

Bodmer, 2016-07-31:
OK, Hopefully I have fixed both issues. I left chip select low at one point in the code and that has now been corrected.

Pin666, 2016-08-01:
Checked it and everything works fine!

Thanks!

closed 2016-08-01 by Pin666