Bodmer/TFT_ILI9341 #7: Only white screen for every test even though serial monitor suggests its working.

lana726, 2016-10-12:
Hi, 
I really need some help with trying to get my TFT ILI9341 LCD screen shield to work with Arduino. 
I have been testing this LCD shield:
http://www.electrodragon.com/product/3-5-320x480-tft-lcd-module-for-arduino-8-bit-parallel/#prettyPhoto
with both my arduino UNO and Mega2560 and neither work. I have used a number of different suggestions from other online forums and the examples here http://www.electrodragon.com/w/3.5%E2%80%98%E2%80%99_320%C3%97480_TFT_LCD_Module_for_Arduino_(8-bit_Parallel) using the Adafruit_GFX and Adafruit_TFTLCD libraries. 

Now trying this library specific for my LCD shield and still no luck. The serial monitor reads and displays everything correctly however nothing appears on the screen. 

Please help, where have I gone wrong. I have made sure all the libraries are loaded correctly etc. Is it a screen fault? However it can run a simple colour test without using any dependent libraries.

[TFT_graphicstest_one_lib_NotWorking.ino.zip](https://github.com/Bodmer/TFT_ILI9341/files/524677/TFT_graphicstest_one_lib_NotWorking.ino.zip)
^ the code 

[image]

^the serial monitor

Any tips would be greatly appreciated.

oldmanegan, 2016-10-12:
Maybe I am missing something here... ILI9341 is SPI, not 8 bit parallel. You seem to be mixing different communication types. If the display is 9341 based, then your sketch appears at a glance to be the right direction, as it is SPI based. However, have you wired the device correctly? Or is it plugged directly into the board as a shield?

Bodmer, 2016-10-12:
The link to the shield indicates you have a display with an ILI9481 controller which is quite different to an ILI9341.

If you had an Adruin Due you could have used my HX8357_Due library which has a configuration option to drive the 8 bit paralell interface display you have, but as you only mention a UNO and Mega you will need to look for a compatible library.

This one may well work:
[https://github.com/prenticedavid/MCUFRIEND_kbv](url)

David Prentice provides support for this library via the Arduino forum under the Displays topic:
[https://forum.arduino.cc/index.php?board=7.0](url)

closed 2016-10-12 by Bodmer

lana726, 2016-10-12:
That's what I've been thinking. Really struggling to figure this out.
The shield plugs directly into the arduino.

Thanks

lana726, 2016-10-12:
Thank you for your help! I shall try those options.