Bodmer/TFT_ILI9341 #9: Include - (minus) in 7 segment font

Ozzy62, 2017-02-11:
First of all, thank you for the nice library, it works a charm.
I use your library to display temperatures, and where I live (Denmark) outside temperatures tend to go below zero in the winter. It would therefore be a great addition (for me) to the library to include the minus sign in the 7 segment font. I do not have the foo to do it my self.

Regards
Svend

Bodmer, 2017-02-24:
Hi, I will see if I can create a new symbol for you this weekend.

Bodmer, 2017-11-16:
Ah, forgot about this one.. is this still a need, or have you sought out another solution?

Ozzy62, 2017-11-19:
Yes, given the time of year, the need has come up again.

Bodmer, 2017-11-24:
I have added the minus in the attached.

[7seg.zip](https://github.com/Bodmer/TFT_ILI9341/files/1502218/7seg.zip)

Can you replace the Font7srle.c file with that attached and let me know if this works for you.

Ozzy62, 2017-11-26:
Hi Bodmer. It works like a charm, thank you very much for your effort. I did have to change: #include <pgmspace.h> to #include <avr/pgmspace.h> but otherwise no problems.

closed 2017-11-26 by Ozzy62

Bodmer, 2017-11-26:
Thanks for fixing that! I will correct it and add to main library.

nico1080, 2022-04-23:
Hi,

It is not included in master.

Bodmer, 2022-04-23:
It is included, [ASCII code 0x2D](https://github.com/Bodmer/TFT_eSPI/blob/master/Fonts/Font72rle.c#L34-L42), so you can use for example:

tft.drawString("-123", 0, 0, 7);

Bodmer, 2022-04-23:
Ah, wrong library! I will add it!

Bodmer, 2022-04-23:
Just checked this library and it is in there, [here](https://github.com/Bodmer/TFT_ILI9341/blob/master/Fonts/Font72rle.c#L36-L44).