Bodmer/TFT_ILI9341 #10: RST, DC, CS set in constructor instead of user.h

avandalen, 2017-10-07:
I prefer to set the RST, DC, CS pins in the constructor instead of user.h. What is your opinion?
I use your library now in the new Spot-welder power controller:
http://www.avdweb.nl/arduino/hardware-interfacing/spot-welder-controller.html

Bodmer, 2017-10-08:
The key advantage of setting them in the header file is that once set correctly all examples that come with the library will run without modification. Since many pin configurations are possible on many different Arduino boards I find this very convenient, both during sketch development for a new project and to test the library on different hardware setups.

closed 2017-10-08 by Bodmer

avandalen, 2017-10-08:
Dear Bodmer,

Yes, in your case this is convenient.

But is has a serious disadvantage. For a project that use many libraries, these libs just have to be installed in the library folder, that’s all. There is nothing that have to edited too by hand. It is not convenient that the users have to edit a user.h file too. Imagine that all libraries have to be edited by hand…
Also, user.h has to be edited again if we update the libraries.

We can overcome this by using an extra constructor, by which we can set the 3 pins, instead of using a user.h file:
So, in case if we want to use the user.h file, we can use the current constructor.
If we want to set the 3 pins in the project file, we can use another constructor:
TFT_ILI9341 tft = TFT_ILI9341(RSRpin, Cspin, DCpin); 
The 3 pins in the user.h file will be ignored than.