Bodmer/TFT_ILI9341 #12: drawBitmap() function dropping pixels off the left edge?

jamsessionein, 2017-11-07:
I apologize, but I could really use some help with this.

I have monochromatic graphics in an array in PROGMEM that I need to display on my screen. This works fine.

```
const unsigned char eyeBitmap1 [5207] PROGMEM = {
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xab, 0xef, 0xef, 0xef, 0x7f, 0xfe, 0xff, 0xdf, 0x54, 
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x55, 0xd5, 0xdf, 0xfd, 0xfd, 0x55, 0xff, 0xbb, 0xfa, 
0x00, 0x00, 0x00, 0x00, 0x00, 0x02, 0xbf, 0xbb, 0xff, 0xdf, 0xff, 0xff, 0xff, 0xd5, 0x54, 
0x00, 0x00, 0x00, 0x00, 0x00, 0x05, 0x57, 0x55, 0x7d, 0x7f, 0xf7, 0xff, 0x75, 0x7f, 0xfe, 
0x00, 0x00, 0x08, 0xae, 0xea, 0xbf, 0xff, 0x7f, 0xff, 0x5f, 0xf7, 0xff, 0xaa, 0xfb, 0xf4, 
0x00, 0x00, 0x05, 0x55, 0x55, 0xdf, 0xff, 0xff, 0xff, 0xfb, 0xfd, 0x55, 0x40, 0x15, 0x5e, 
0x00, 0x00, 0xab, 0xbf, 0xbf, 0xff, 0xff, 0xdf, 0xff, 0xff, 0xea, 0xaa, 0x20, 0x02, 0xde, 
0x00, 0x01, 0x55, 0x7f, 0x7f, 0x7f, 0xf7, 0x7f, 0x7f, 0xf7, 0xd5, 0x40, 0x00, 0x00, 0x56, 
0x00, 0x2e, 0xff, 0xff, 0xf7, 0xff, 0xff, 0xff, 0xf7, 0xff, 0xfe, 0x80, 0x00, 0x00, 0xae, 
0x00, 0x15, 0x7f, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xfd, 0x54, 0x00, 0x00, 0x00, 0x04, 
0x00, 0x3b, 0xff, 0xfd, 0x55, 0x55, 0xdf, 0xdd, 0x57, 0xfb, 0xa0, 0x00, 0x00, 0x00, 0x02, 
0x00, 0x75, 0x7f, 0xff, 0xfe, 0xff, 0xff, 0xea, 0xff, 0xf5, 0x40, 0x00, 0x00, 0x00, 0x00, 
0x02, 0xff, 0xf5, 0x57, 0x55, 0x55, 0x55, 0x01, 0x55, 0x7a, 0x80, 0x00, 0x00, 0x00, 0x0a, 
0x05, 0xfd, 0xff, 0xbe, 0xaa, 0xa2, 0xaa, 0xab, 0xff, 0xd4, 0x00, 0x00, 0x00, 0x00, 0x00, 
0x2b, 0xff, 0xd5, 0x55, 0x40, 0x45, 0x51, 0x45, 0xdf, 0xaa, 0x20, 0x00, 0x00, 0x00, 0x00, 
0x15, 0xff, 0xee, 0xef, 0xea, 0xaa, 0xaa, 0xae, 0xf5, 0x10, 0x00, 0x00, 0x00, 0x00, 0x00, 
0xff, 0xd5, 0x55, 0x55, 0x51, 0x11, 0x51, 0x77, 0xea, 0x88, 0x80, 0x00, 0x00, 0x00, 0x00, 
0x5d, 0xfa, 0xba, 0xaa, 0xaa, 0xaa, 0xab, 0xfd, 0x54, 0x40, 0x00, 0x00, 0x00, 0x00, 0x00, 
0xff, 0xd5, 0x55, 0x55, 0x00, 0x40, 0x55, 0xfb, 0xaa, 0xaa, 0x00, 0x00, 0x00, 0x00, 0x00, 
0x7f, 0xee, 0xea, 0xea, 0xaa, 0xab, 0xef, 0xd5, 0x14, 0x00, 0x04, 0x00, 0x00, 0x00, 0x00, 
0xff, 0x55, 0x15, 0x50, 0x15, 0x15, 0x5f, 0xfe, 0xea, 0xaa, 0x28, 0x80, 0x00, 0x00, 0x00, 
0xff, 0xaa, 0xab, 0xaa, 0xbb, 0xff, 0xfd, 0x54, 0x44, 0x54, 0x00, 0x00, 0x00, 0x00, 0x00, 
0xfd, 0xd5, 0x55, 0x55, 0x55, 0xdd, 0xff, 0xbb, 0xa2, 0x88, 0x80, 0x00, 0x00, 0x00, 0x00, 
0x7f, 0xea, 0xaa, 0xef, 0xff, 0xff, 0x75, 0x55, 0x55, 0x50, 0x10, 0x00, 0x00, 0x00, 0x00, 
0xf5, 0x55, 0x55, 0x57, 0xff, 0xff, 0xff, 0xef, 0xfe, 0xfa, 0x8e, 0x88, 0x00, 0x00, 0x00, 
0xff, 0xbb, 0xff, 0xfd, 0x5f, 0xdd, 0xdd, 0x5d, 0xcf, 0xdc, 0xdc, 0x40, 0x00, 0x00, 0x00, 
0xfd, 0x5f, 0xff, 0xff, 0xbf, 0xfb, 0xff, 0xff, 0xdf, 0x9d, 0xfb, 0x20, 0x00, 0x00, 0x00, 
0x7f, 0xff, 0x7f, 0xd5, 0x55, 0x75, 0x7f, 0xff, 0xfe, 0xff, 0x74, 0xe4, 0x00, 0x00, 0x00, 
0xff, 0xfe, 0xee, 0xaa, 0xff, 0xff, 0x55, 0x11, 0x15, 0x55, 0x9c, 0x6e, 0xe8, 0x00, 0x00, 
0x55, 0xd5, 0x55, 0x55, 0x5d, 0x7f, 0xaa, 0xaa, 0xaa, 0xaa, 0x38, 0xd1, 0x00, 0x40, 0x00, 
0xab, 0xaa, 0xaa, 0xaa, 0xbf, 0xfd, 0x40, 0x00, 0x00, 0x04, 0x10, 0x9e, 0xa2, 0x80, 0x00, 
0x15, 0x01, 0x11, 0x11, 0x57, 0xf8, 0x00, 0x00, 0x00, 0x08, 0x20, 0xfb, 0x51, 0x00, 0x00, 
0xaa, 0xaa, 0xae, 0xaa, 0xff, 0x40, 0x00, 0x00, 0x00, 0x00, 0x01, 0x1f, 0xec, 0x0c, 0x00, 
0x00, 0x05, 0x00, 0x55, 0xde, 0x20, 0x00, 0x00, 0x00, 0x00, 0x02, 0xfb, 0xd6, 0x14, 0x00, 
0x20, 0x0a, 0xaa, 0x2b, 0xb0, 0x00, 0x04, 0x40, 0x04, 0x00, 0x00, 0x15, 0xff, 0xa2, 0x00, 
0x00, 0x01, 0x01, 0x15, 0x6a, 0xaa, 0xae, 0xa8, 0xa8, 0xa0, 0x00, 0xab, 0xff, 0xd0, 0x00, 
0x00, 0x88, 0xa8, 0xaf, 0x0d, 0xd5, 0x45, 0x79, 0x84, 0xc0, 0x00, 0x00, 0x55, 0xfa, 0x82, 
0x00, 0x00, 0x00, 0x54, 0xb9, 0x8f, 0xfc, 0xe4, 0x62, 0x80, 0x28, 0x02, 0x3f, 0xdd, 0x40, 
0x00, 0x02, 0x2a, 0xa9, 0x7f, 0xfb, 0xda, 0xc6, 0x10, 0x00, 0x2c, 0x00, 0x15, 0xfd, 0xaa, 
0x00, 0x00, 0x05, 0x76, 0xfb, 0xfd, 0x77, 0x00, 0x00, 0x00, 0x00, 0x00, 0x2b, 0xec, 0x00, 
0x00, 0x0a, 0x00, 0xb6, 0xef, 0xef, 0x7b, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x67, 0xe2, 
0x00, 0x00, 0x05, 0x6c, 0x5b, 0x11, 0x3c, 0x00, 0x00, 0x00, 0x00, 0x00, 0x02, 0xb4, 0x50, 
0x00, 0x00, 0x22, 0xf9, 0xf2, 0x7a, 0xe0, 0x00, 0x04, 0x00, 0x00, 0x00, 0x00, 0x7c, 0x8a, 
0x00, 0x00, 0x01, 0xd5, 0x71, 0x31, 0x08, 0x00, 0x22, 0x80, 0x00, 0x00, 0x20, 0x0d, 0x56, 
0x00, 0x0a, 0x2b, 0x6a, 0xae, 0xfc, 0x00, 0x00, 0xc0, 0x00, 0x00, 0x00, 0x00, 0x15, 0x1a, 
0x00, 0x00, 0x07, 0x44, 0x54, 0x4a, 0x00, 0x00, 0x00, 0x00, 0x20, 0x00, 0x82, 0xa0, 0x14, 
0x00, 0x02, 0x2e, 0x80, 0x2a, 0xc0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x15, 0x5e, 
0x00, 0x00, 0x1d, 0x10, 0x16, 0xa0, 0xa8, 0xaa, 0x80, 0x00, 0x08, 0x00, 0x00, 0x03, 0x54, 
0x00, 0x00, 0xb8, 0x40, 0xa6, 0x08, 0x11, 0x15, 0x00, 0x00, 0x00, 0x00, 0x00, 0x05, 0x7e, 
0x00, 0x00, 0x38, 0x05, 0xda, 0x01, 0x1f, 0xaf, 0xa2, 0x00, 0x00, 0x00, 0x00, 0x0b, 0xfc, 
0x00, 0x00, 0x49, 0x9a, 0x10, 0x40, 0x0d, 0xd7, 0x40, 0x00, 0x00, 0x00, 0x10, 0x05, 0xfe, 
0x00, 0x00, 0x40, 0x15, 0xe0, 0x08, 0x02, 0xfe, 0xe0, 0x88, 0x00, 0x00, 0x04, 0x2b, 0xfe, 
0x00, 0x02, 0xa0, 0xdf, 0x06, 0x80, 0x0f, 0x57, 0x00, 0x00, 0x00, 0x00, 0x00, 0x15, 0x7e, 
0x00, 0x01, 0x41, 0x03, 0x22, 0x00, 0x05, 0xed, 0x80, 0x00, 0x00, 0x20, 0x10, 0x2b, 0xfc, 
0x00, 0x03, 0x1b, 0x77, 0x5a, 0x00, 0x0b, 0x5f, 0xd0, 0x00, 0x00, 0x00, 0x13, 0x05, 0xfe, 
0x00, 0x00, 0x04, 0x59, 0xa0, 0x00, 0x05, 0xf6, 0xc0, 0x08, 0x80, 0x28, 0x03, 0xe0, 0xf4, 
0x00, 0x04, 0x8b, 0x87, 0x00, 0x00, 0x03, 0xef, 0xc0, 0x91, 0x00, 0x00, 0x00, 0x57, 0x50, 
0x00, 0x0c, 0x48, 0x69, 0x80, 0x00, 0x05, 0x59, 0xc0, 0x2b, 0xa0, 0x28, 0x04, 0x2f, 0xf4, 
0x00, 0x1a, 0xb5, 0xbd, 0x80, 0x00, 0x02, 0xff, 0x88, 0x3d, 0x50, 0x44, 0x01, 0x00, 0x00, 
0x00, 0x04, 0x00, 0x5c, 0x80, 0x00, 0x01, 0x50, 0x40, 0x3f, 0xfa, 0x6a, 0x80, 0x00, 0xf6, 
0x80, 0x28, 0x0a, 0xf8, 0x00, 0x00, 0x00, 0xaa, 0x84, 0xe1, 0x74, 0x51, 0x05, 0x00, 0x00, 
0x00, 0x00, 0x15, 0x51, 0x00, 0x00, 0x00, 0x55, 0x24, 0x01, 0xfe, 0x7a, 0xa3, 0xa0, 0x0e, 
0x02, 0x00, 0x2a, 0xa2, 0x00, 0x00, 0x00, 0x6e, 0x06, 0x19, 0xfc, 0x75, 0x51, 0x9f, 0xfa, 
0x00, 0x40, 0x05, 0x60, 0x00, 0x00, 0x00, 0x16, 0x54, 0x00, 0x7e, 0xda, 0xee, 0xd9, 0x54, 
[... 34 lines cut ...]
0x00, 0x00, 0x8a, 0xae, 0xaa, 0x80, 0x00, 0x00, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 
0x00, 0x00, 0x41, 0x00, 0x55, 0x40, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
0x00, 0x08, 0x08, 0x20, 0xba, 0xa8, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
0x00, 0x00, 0x00, 0x04, 0x10, 0x40, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
0x00, 0x00, 0x00, 0x22, 0x30, 0x22, 0x00, 0x20, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
0x00, 0x00, 0x00, 0x40, 0x04, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
0x00, 0x00, 0x00, 0x00, 0x82, 0x80, 0x80, 0xc0, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
0x00, 0x00, 0x00, 0x04, 0x00, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
0x00, 0x00, 0x00, 0x00, 0x00, 0x20, 0x20, 0x20, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
0x00, 0x00, 0x08, 0x00, 0x20, 0x0a, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
};
```

The above array was generated from [this image](https://i.imgur.com/vbUvDNB.png). 

I was initially using the Adafruit ILI9341 library, but I was getting frustrated with the slow speed of the drawBitmap() function they provided. I switched over to using this library and was immediately thrilled with how quick the images I was working with would populate onto the screen. However, I'm having a bit of a problem in that I'm getting "noise" on the left side of the bitmaps displayed.

[An image is the best way to demonstrate the issue](https://i.imgur.com/2XQPLbo.png). 

Running on the same hardware, same screen, same wiring, and it's the same array being fed into both functions, so I'm not sure where the discrepancy lies, but I'm afraid it might be something in the drawBitmap() function itself, which is so far beyond my ability to understand that it's embarassing. 

In my User_Setup.h, I've got #define FAST_GLCD and #define FAST_LINE uncommented, my pins are defined correctly, and I've commented out all of the fonts save for the initial #define LOAD_GLCD. #define CLIP_CHECK and #define SUPPORT_TRANSACTIONS are presently commented. 

Any idea what I'm doing wrong? I'm an amateur at best here, and I honestly have no idea how to read useful data out of the byte array above to figure out if there's some formatting thing I have wrong in there that would cause it to work fine with one library's functions but not another's.

jamsessionein, 2017-11-07:
[image]

Another example. This one is a set of 3 arrays overlaid on one another with 3 drawBitmap() calls, one for each. Same kind of results - something is up with that left edge and I don't know what.

jamsessionein, 2017-11-08:
Another note. I was thinking maybe the issue had to do with the size of the image I'm using, so I approached it with something much smaller.

```
const unsigned char iconLAPD [] PROGMEM = {
0x49, 0x00, 0x20, 0x30, 0x03, 0x03, 0xe1, 0xe0, 
0x7d, 0xe7, 0xff, 0x30, 0x03, 0x03, 0xf1, 0xf0, 
0xd7, 0x44, 0x01, 0x30, 0x07, 0x83, 0x39, 0x98, 
0x74, 0x81, 0xd6, 0x30, 0x07, 0x83, 0x19, 0x98, 
0x0f, 0x67, 0x4a, 0x30, 0x04, 0x83, 0x19, 0x8c, 
0x00, 0x02, 0x87, 0x30, 0x0c, 0xc3, 0x71, 0x8c, 
0xff, 0xe7, 0x70, 0x30, 0x0d, 0xc3, 0xe1, 0x8c, 
0x00, 0x00, 0x00, 0x30, 0x0f, 0xc3, 0x01, 0x98, 
0x7f, 0xc3, 0xfe, 0x30, 0x18, 0x63, 0x01, 0x98, 
0x40, 0x41, 0x24, 0x3f, 0x98, 0x63, 0x01, 0xf0, 
0x7f, 0xc6, 0x62, 0x1f, 0x18, 0x63, 0x01, 0xe0
};
```
The above is meant to produce this little guy:

[image]

And look what happens when I flip between libraries:

[image]

Now, is one pixel in that whole array a big deal? No way. Most people wouldn't even notice that, or would assume it's intentional. But the problem I have is that this is an unknown error that compounds - the more I draw, the more I end up deviating from what I intend to get on the screen. 

I'm honestly not even sure how to debug this kind of issue, though.

Bodmer, 2017-11-16:
I suspect this is a timing issue related to the clock rate and power supply voltage. What is your setup and how is it powered?

Another potential issue is that the SPI clock is running at 6MHz so perhaps the delay based SPI approach (used for speed) is not compatible with the lower clock rate?

Bodmer, 2017-11-24:
I had a look at the library code and see it uses the [fastPixel function](https://github.com/Bodmer/TFT_ILI9341/blob/master/TFT_ILI9341.cpp#L1089-L1117) to draw the bitmap.

If you are able you could try slowing it down by replacing the spiWait12(); with spiWait14(); and spiWait14(); with spiWait15(); that may do the trick. Let me know what works.

closed 2018-08-14 by Bodmer