Bodmer/TFT_ILI9341 #13: New fonts?

MohammadMubin, 2017-11-25:
Hi Thanks for this amazing library but the one thing this library is lacking on is custom fonts. i have tried the adafruit font generator but no avail. and Some tools from the UTFT library and still no luck. Any help would be great!

(and no i am not making a bomb :P)

Regards,
M.Mubin

Bodmer, 2017-11-25:
Hi, as you surmised this library does not support custom fonts at the moment. This is for historical reasons that you can read about [here](https://www.instructables.com/id/Arduino-TFT-display-and-font-library/).
Later Adafruit fonts to the library but they are very slow to render and so I did not add them. In later libraries for the faster parallel displays I found rendering speed became acceeptable so I added them.

My favourite Arduino IDE compatible process is now the ESP8266 so all my recent development work is centred on the library [here](https://github.com/Bodmer/TFT_eSPI), which allows custom fonts to be added.

I will add your request for custom fonts to my "to do" list but I do not have much spare time at the moment.

MohammadMubin, 2017-11-25:
Hmmm, So the Parallel 8 bit displays are faster then the SPI? from my raspberry pi/arm knowledge that was false. i have tonnes of those displays but only one with suitable driver works (ILI9488).
and ESP8266??? i mean those don't even have more then 2 GPIOS well the -01's but still i mean i got dozen of those too laying around with no use (the 01's not the 12E)
 

And thanks for adding my request in your "to do" list  :) Greatly Appreciated!
//  my English is not that great.