Bodmer/TFT_ILI9341 #15: Need new sign for Font7

P123GT, 2018-03-21:
Hi,
I need the "-" (2D) sign at the Font7srle. I have seen it at the lib for the actual eSPI and simply added the code 
`// Make - sign look like a segment
PROGMEM const unsigned char chr_f7s_2D[] = 
{
0x7F, 0x7F, 0x7F, 0x7F, 0x7F, 0x27, 0x8E, 0x0E, 
0x92, 0x0A, 0x96, 0x09, 0x94, 0x0C, 0x90, 0x7F, 
0x7F, 0x7F, 0x7F, 0x7F, 0x47
};
`
from there to the Font7srle.c of the TFT_ILI9341 but it still doesn't work.
Is there something else to change or could you do that?

René

Bodmer, 2018-03-22:
Use the font file from here:

https://github.com/Bodmer/TFT_eSPI/tree/master/Fonts

That one has the minus sign added.

P123GT, 2018-03-23:
Great, thanks!

closed 2018-03-23 by Bodmer