Bodmer/TFT_ILI9341 #17: CS Pin should not be HIGH right after sending the command byte

abhra0897, 2019-10-06:
You've made CS pin HIGH right after issuing a command. But as per the datasheet, the pin should be low until all the command data (followed by the command) are sent. For example, if the command is 0x03 and then we need to send 2 bytes of data 0x4F and 0x81, the CS line will be pulled low before sending 0x03 and can only be high after 0x81.
Can you please help to clear my doubt? thanks.
```
void TFT_ILI9341::writecommand(uint8_t c)
{
  savedSPCR = SPSR;// We need this here for some reason...
  TFT_DC_C;
  TFT_CS_L;
  spiwrite(c);
  //Should it be here?
  TFT_CS_H;
}
```

[image]

Bodmer, 2019-10-06:
It's OK read p35 of the [datasheet](https://www.crystalfontz.com/controllers/Ilitek/ILI9341/).

abhra0897, 2019-10-07:
Got it. Actually I'm writing a parallel 8-bit driver (8080-II) for STM32 based on your library. But your's is SPI based. For parallel interface, CS must be high as per the above screenshot. For serial though, CS is allowed to go high between cmd & data or data & data (as per p35). Thanks for replying quickly.

closed 2019-10-07 by abhra0897

Bodmer, 2019-10-07:
See p42 and p43, CS can go high between bytes even in parallel mode.