Bodmer/TFT_ILI9341 #20: ATmega128 not supported ?

Greece2, 2020-12-09:
As the lib is using the hardware-SPI, it seems it is not working with the ATmega128.
The SPI-pins are:
MOSI : PB2 -> 10
MISO: PB3 -> 11
SCK: PB1 -> 9
SS: PB0 -> 8

Any chance to integrate that into the lib?  Or can one use the "User_Setup.h" to define these pins?  Or the "fast_pin.h" ?

Best regards and thanks for the support !

Here part of the compiler's output:

> \libraries\TFT_ILI9341-master/TFT_FastPin.h:479:2: warning: #warning "No pin/port mappings found, pin access will be slightly slower.  See fastpin.h for info." [-Wcpp]
 #warning "No pin/port mappings found, pin access will be slightly slower.  See fastpin.h for info."

Bodmer, 2020-12-10:
I suspect you are right in that the fast pin macros do not support that processor.

Greece2, 2020-12-10:
Could that be fixed?  The Mega64, 1281, 2561 have the same pinout.  Thanks !

Greece2, 2020-12-19:
No chance at all ?  
The lib is such a great tool !  Well, would be ...

Bodmer, 2020-12-19:
The library uses fastpin, this template library is from FastLED http://fastled.io

I do not know what changes would be needed but it could be that the FastLED library has been updated to include the ATmega128?

Greece2, 2020-12-20:
Thanks for your feedback, I'll check that out.

Have a good time and stay safe !

aliihsanhavan, 2021-05-03:
@Greece2 if you are using megacore, add below section to TFT_Fastpin.h file and you are good to go.

[atmega128_Pinout.txt](https://github.com/Bodmer/TFT_ILI9341/files/6415024/atmega128_Pinout.txt)