Bodmer/TFT_ILI9341 #23: Help, please.

udas2003, 2021-05-25:
I just only begin to work with SPI displays and study the examples.
I examined some examples and no one does work, just white screen, that periodicaly turn to dark blue.
At  the same time examples  "graphictest" from Adafruit_ILI9341 and "mock_ili9341"  from Adafruit-GFX
work as should.
At all cases I use the same set for Arduino pins.
Setup - UNO, converter 5v to 3.3v , chinese "red" SPI display 3.2" 240x320 from Ali.
Igor

Bodmer, 2021-05-25:
Run the Read_User_Setup sketch and check the output is as you expect for your hardware.

udas2003, 2021-05-26:
Hello Bodmer!

I’m not skilled User at all!

Retired, 69 years old – learn Arduino as a hobby.

So, would you be so kind to give me more detailed instructions?

I did find only one  “Read_User_Setup sketch”  in your TFT_eSPI library.

I looked over the sketch and, as I understand, the one is for 32 bits processors.

But I have modest UNO.

Is this sketch what you recommend me?

Best wishes

Igor

 

 

 

 

Run the Read_User_Setup sketch and check the output is as you expect for your hardware.

—

Bodmer, 2021-05-26:
Sorry, my response was for a different library. Currently I am travelling but will post back in a few days.

udas2003, 2021-05-26:
Sorry, my response was for a different library. Currently I am travelling but will post back in a few days.

Wish fun days!

udas2003, 2021-05-26:
See this link. You must use voltage level converters as described.
https://www.google.com/amp/s/www.instructables.com/Arduino-TFT-display-and-font-library/%3famp_page=true

—
Ok, at holydays I”ll buy resistors and check such converter.

Now, in successful sketches  from Adafruit_ILI9341 and Adafruit_GFX libraries I used

*	3.3V to display pin 1 (VCC) 
*	3.3V to display pin 8 (LED) 
*	UNO 0V (GND) to display pin 2 (GND) 
*	UNO pins 7,9(8???),10,11,13 through the converter on TXS0108E chip (see above).

The pins numbers were chosen from  TFT_Clock scetch or User_Setup.h:

“// These are the control pins I use on my UNO/Nano/Pro Micro/ATmega328 setup

//   TFT_CS  10  // Chip select control pin

//   TFT_DC   9  // Data Command control pin

//   TFT_RST  7  // Reset pin (could connect to Arduino RESET pin)”

In User_Setup.h :

I commented out all Fonts besides Font4 (for TFT_Clock sketch),

In “speed up options”  only  #define FAST_LINE is uncommented.

I don’t see  “Other F_AS_T option”

“ If using an UNO or Mega (ATmega328 or ATmega2560 processor) then for best

performance use the F_AS_T option found in the User_Setup.h file in the

TFT_ILI9341 library folder.” Or do you mean #define FAST_LINE?

 

Best wishes

Igor