Bodmer/TFT_ILI9341 #26: Where has the example Extended_ASCII_Demo gone?

PaulskPt, 2021-11-04:
Hi, I am struggling with getting just a small group of 6 letters of the Latin-5 group ISO/IEC8859-9 (Turkish) letters displayed. I wanted to see an example. While googling on internet I came across your answer on this Arduino forums page:
https://forum.arduino.cc/t/solved-for-the-moment-how-to-display-latin-characters-on-tft/580962/2
In your answer you gave this link: 
https://github.com/Bodmer/Adafruit_ILI9341/blob/master/examples/Extended_ASCII_Demo/Extended_ASCII_Demo.ino
The link no longer exists. I followed the root of that link; arrived in this repo. The example sketch is not in the current list of examples.  Do you have that example still somewhere? Thank you.

Bodmer, 2021-11-05:
The sketch was part of a proposed change to the Adafruit_GFX library that was not adopted so it got deleted.

I suspect this is the sketch:
[Extended_ASCII_Demo.zip](https://github.com/Bodmer/TFT_ILI9341/files/7488820/Extended_ASCII_Demo.zip)

closed 2021-11-05 by Bodmer

PaulskPt, 2021-11-05:
Thanks Bodmer !