Bodmer/TFT_ILI9341 #27: Pacman Example with Issues

hasenradball, 2022-01-05:
HI Bodmer,

I created an Pacman Example with an ILI9341, but it will not run very well.
Do you maybe have an idea, what could be the issue here?

I copy the code here:

```
/***************************************************
  ESP8266 PacMan Example on the Wemos mini D1
 ****************************************************/

#include <TFT_eSPI.h>
#include <SPI.h>

TFT_eSPI tft = TFT_eSPI();

/*
// for Wemos D1 Mini use CS->16; DC->15
#define TFT_CS 16
#define TFT_DC 15
#define TFT_MOSI 13
#define TFT_CLK 14
#define TFT_RST -1
#define TFT_MISO 12
*/

#define TFT_LED 5

void pacman(){
  int dir {0};
  int m {0};
  tft.fillScreen(ILI9341_BLACK);
  for (int x=-16, y = 15; x < 320; ++x) {
    tft.fillScreen(ILI9341_BLACK);
    tft.fillCircle(x, y, y, ILI9341_YELLOW);
    tft.fillCircle(x+1, 7, 2, ILI9341_BLUE);
    tft.fillTriangle(x, 15, x+15, 15-m, x+15, 15+m, ILI9341_BLACK);
    if(m!= 0) {
      tft.fillTriangle(x, 15, x+15, 15-m, x+15, 15+m, ILI9341_BLACK);
    }
    if(dir == 0) {
      m++;
      if(m == 15) {
        dir = 1;
      }
    }
    if(dir == 1) {
      m--;
      if(m == 0) {
        dir = 0;
      }
    }
  }
  delay(4);
}

void setup() {
  Serial.begin(115200);
  Serial.println("\nILI9341 Pacman-Test!"); 
  
  pinMode(TFT_LED, OUTPUT);
  digitalWrite(TFT_LED, LOW);
  
  tft.begin();
}

void loop(void) {

  tft.setRotation(3);
  pacman();
}
```

Bodmer, 2022-01-06:
The problem is that clearing the screen takes a significant time sinnce 320*240 =76,800 pixels have to be written.

Since the paccman only moves 1 pixel at a time you only need to clear a circle 1 pixel bigger:
```
void pacman(){
  int dir {0};
  int m {0};
  tft.fillScreen(ILI9341_BLACK);
  for (int x=0, y = 15; x < 300; ++x) {
    //tft.fillScreen(ILI9341_BLACK);
    tft.drawCircle(x, y, y+1, ILI9341_BLACK);
    tft.fillCircle(x, y, y, ILI9341_YELLOW);
    tft.fillCircle(x+1, 7, 2, ILI9341_BLUE);
    tft.fillTriangle(x, 15, x+15, 15-m, x+15, 15+m, ILI9341_BLACK);
    if(m!= 0) {
      tft.fillTriangle(x, 15, x+15, 15-m, x+15, 15+m, ILI9341_BLACK);
    }
    if(dir == 0) {
      m++;
      if(m == 15) {
        dir = 1;
      }
    }
    if(dir == 1) {
      m--;
      if(m == 0) {
        dir = 0;
      }
    }
    delay(20);
  }
}
```
Ideally however it would be better to draw the pacman into a small sprite and then put that on the screen. See the Sprite->Sprite_draw example.

hasenradball, 2022-01-06:
Hi,

thank you very much for your answer.
It works great!
Thanks
:-)

closed 2022-01-06 by hasenradball